beagleboard/librobotcontrol #161: Unable to use PRU to send PMW signals to esc

duwke, 2019-05-11:
**Describe the bug**
Using an older version of debian, I was able to use the following command to spin a brushless motor:
sudo rc_test_escs -c 1 -s .2
I started over with an ubuntu image, using the same hardware, and the same command now doesn't do anything.

rc_test_drivers:
```
Kernel: 4.14.108-ti-r104
BeagleBoard.org ROS Image 2019-04-28
Debian: buster/sid

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4
`

sudo /opt/scripts/tools/version.sh:
`
git:/opt/scripts/:[c2a5ae96746d10d711c60bc84ba244f4cd7b75c3]
eeprom:[A335BNLTBLA21717EL001394]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org ROS Image 2019-04-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc8b5ad3a1f]:[location: dd MBR]
kernel:[4.14.108-ti-r104]
nodejs:[v8.10.0]
device-tree-override:[dtb=am335x-boneblue.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190429.0-0rcnee0~bionic+20190429]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~bionic+20190227]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~bionic+20190327]
groups:[beagle : beagle adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep remote
[    1.417683] remoteproc remoteproc0: wkup_m3 is available
[    1.506267] remoteproc remoteproc0: powering up wkup_m3
[    1.506393] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.510329] remoteproc remoteproc0: remote processor wkup_m3 is now up
[    9.863376] remoteproc remoteproc1: 4a334000.pru is available
[    9.879939] remoteproc remoteproc2: 4a338000.pru is available
[  110.318362] remoteproc remoteproc2: powering up 4a338000.pru
[  110.321319] remoteproc remoteproc2: Booting fw image am335x-pru1-rc-servo-fw, size 35164
[  110.321360] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  119.730366] remoteproc remoteproc2: stopped remote processor 4a338000.pru
[... 78 lines cut ...]
[  754.873759] remoteproc remoteproc2: powering up 4a338000.pru
[  754.878246] remoteproc remoteproc2: Booting fw image am335x-pru1-rc-servo-fw, size 35164
[  754.878293] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  762.639892] remoteproc remoteproc2: stopped remote processor 4a338000.pru
[  775.243588] remoteproc remoteproc2: powering up 4a338000.pru
[  775.243870] remoteproc remoteproc2: Booting fw image am335x-pru1-rc-servo-fw, size 35164
[  775.243911] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  782.592204] remoteproc remoteproc2: stopped remote processor 4a338000.pru
[  782.592467] remoteproc remoteproc2: powering up 4a338000.pru
[  782.597672] remoteproc remoteproc2: Booting fw image am335x-pru1-rc-servo-fw, size 35164
[  782.597721] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  786.537709] remoteproc remoteproc2: stopped remote processor 4a338000.pru
dmesg | grep pinctrl-single
[    1.069850] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.116293] pinctrl-single 44e10800.pinmux: pin PIN29 already requested by 481a8000.serial; cannot claim for ocp:helper
[    1.127315] pinctrl-single 44e10800.pinmux: pin-29 (ocp:helper) status -22
[    1.134264] pinctrl-single 44e10800.pinmux: could not request pin 29 (PIN29) from group pins  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`
Using ubuntu because it has debian ROS support, but I guess I can always go back to the base debian image if need be.

Any suggestions are appreciated.

RobertCNelson, 2019-05-11:
@duwke, I see a possible conflict:

```
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
```

```
dmesg | grep pinctrl-single
[    1.069850] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.116293] pinctrl-single 44e10800.pinmux: pin PIN29 already requested by 481a8000.serial; cannot claim for ocp:helper
[    1.127315] pinctrl-single 44e10800.pinmux: pin-29 (ocp:helper) status -22
[    1.134264] pinctrl-single 44e10800.pinmux: could not request pin 29 (PIN29) from group pins  on device pinctrl-single
```

Please drop that change in /boot/uEnv.txt and retry again..

Regards,

duwke, 2019-05-12:
Kind internet stranger, you have made my day.  I've been poking on this for two weeks after work, and this is the last little token that got ardurover working.  Bless you!

closed 2019-05-12 by duwke