beagleboard/librobotcontrol #169: BeagleBone Black - Tiva C connection over SPI

hjmcu, 2019-10-03:
I'm trying to connect a  BeagleBone Black with a Texas Instrument Tiva C using SPI . On the master BBB I'm using rc_spi functions to configure the SPI pins and send 8 bytes of data to the slave Tiva C. The 8 bytes are sent to the Tiva C (checked using a Logic Analyzer) but the Tiva C only receive the first byte. Program on the Tiva C works fine (Checked using Tiva C to Tiva C connection) 
The communication of the BBB with the peripheral device (Tiva C) is Freescale SPI mode 0 
The pinout on the BBB is:
GND
3.3V
MOSI (P9_30)
MISO (P9_29)
SCK (P9_31)
On the BBB the Slave Select/FSS/CS is always at "0" when transmitting data frames 
In this configuration the Tiva C needs SSInFss to return to "1" every time a frame is sent
(SSInFss is forced High)
Is there a way to force the Slave Select pin on the BBB from "0" to "1" when a frame is Tx  or is that the way the SPI Linux driver or rc_spi library works?

StrawsonDesign, 2019-10-03:
This is why I set up the slave select pin to be configurable in manual mode
since the Linux spi driver often doesn't behave as we want. I suggest
trying the manual mode option and toggling the slave select pin on and off
in your software where appropriate.  Good luck!

hjmcu, 2019-10-03:
I will try the manual mode, thank you

closed 2019-10-03 by hjmcu