beagleboard/librobotcontrol #173: Support for the Beaglebone AI

blkist, 2019-11-30:
Hello, I have been trying to get the Robotics Cape working on the Beaglebone AI. As I understand it, the device tree that the RC control library uses, is not able to be used with the AM5729 that the BBAI has. Before I say anything more, let me say that I have only basic programming skills and am no electronics engineer by any means, so I honestly have no idea how much work would go into this. I just wanted to use some of my programs I wrote for the BeagleBone Blue on the BBAI. Thanks.

StrawsonDesign, 2019-11-30:
I believe there is already a device tree dtbo for the robotics cape in the
BBAI image. Which functions in particular are you looking to use? I started
a branch early on when the BBAI was still in development but didn't get
very far.

blkist, 2019-11-30:
Well, whenever I execute *rc_battery_monitor*, with the latest BBAI image,
this is the message I get:
"rc_battery_monitor can only run on BB Blue, Black, and Black wireless when
the roboticscape device tree is in use."

I tried running the device tree *configure_robotics_dt.sh*, which I found
here:
https://github.com/StrawsonDesign/librobotcontrol/blob/master/device_tree/configure_robotics_dt.sh
and from that point, it would no longer boot.

I'm not sure if there's something relatively simple that can be done or
not.
As far as what functions in particular I'd like to use, I guess all of the
functionality of the Robotics cape in general. My program uses the GPS
UART, the motor drivers, and a couple I2C.

blkist, 2019-12-26:
Hello, I am not trying to be a bother, but it there any way you could point me in the right direction. Still working on getting the Robotics Cape device tree overlay to work with the BBAI.

jadonk, 2020-06-03:
Work is started: https://github.com/beagleboard/librobotcontrol/commit/99a08dbd5fc133c4b199859a849d8b1ed5a7ee64

jadonk, 2020-06-03:
FYI, this is scheduled for the v2020.08 BeagleBoard.org Debian image, though you should be able to pull in on-going work ahead of them with some extra work.

blkist, 2020-06-03:
That makes me super excited!! Thanks for the update!!

jstumfoll, 2020-07-04:
@jadonk I pulled in the v1.1 branch and built everything. Trying to get it to work on my BBAI but seems something is still missing. The first time I tried to install the configure_robotics_dt.sh script updated the uEnv.txt and it wouldn't boot. I loaded the backup file and tried a second time, skipping that step, and it seems to kind of work.  When I ran the rc_test_drivers I get more errors than passing tests though. Looking at the configure_robotics_dt script and files I'm a little confused. It seems that an overlay for the BBAI still needs to be created, however the readme text says "The overlay is deprecated now." Does that mean these overlays aren't used anymore? What could be causing the issue with the library drivers?

debian@beaglebone:~/src/librobotcontrol/examples/bin$ rc_test_drivers

Kernel: 4.14.108-ti-r136
BeagleBoard.org Debian Buster IoT TIDL Image 2020-04-06
Debian: 10.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm0
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
PASSED: pru-rproc
ERROR:  uart1 driver not loaded
ERROR:  uart2 driver not loaded
ERROR:  uart4 driver not loaded
PASSED: uart5
ERROR:  i2c1 driver not loaded
ERROR:  i2c2 driver not loaded
ERROR: spi driver not loaded
ERROR:  LED driver not loaded
PASSED: ADC iio

Currently running on a:
MODEL_BB_AI
Robot Control library Version:
1.1.0

jstumfoll, 2020-07-05:
Update: I found am5729-beagleboneai-roboticscape.dtb and updated the configure_robotics_dt.sh script with a few lines of code:

```
TREE_BBAI_RC=am5729-beagleboneai-roboticscape.dtb

# test for BBAI
elif   [ "$MODEL" == "BeagleBoard.org BeagleBone AI" ]; then

	# if the roboticscape tree is available, use that
	if [ -a "/boot/dtbs/$UNAME/$TREE_BBAI_RC" ]; then
		DTB="$TREE_BBAI_RC"
	else
		echo "ERROR, can't find $TREE_BBAI_RC for this kernel."
		echo "no changes made to uEnv.txt"
	fi

```

This fixed the issue with some of the drivers, however the PWM and EQEP still doesn't work:

debian@beaglebone:~/src/librobotcontrol/examples/bin$ rc_test_drivers

Kernel: 4.14.108-ti-r136
BeagleBoard.org Debian Buster IoT TIDL Image 2020-04-06
Debian: 10.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm0
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
ERROR:  i2c1 driver not loaded
ERROR:  i2c2 driver not loaded
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_AI
Robot Control library Version:
1.1.0

jadonk, 2020-07-19:
For the Linux bus drivers, we are fixing these as part of a GSoC project on cape compatibility. Once the interfaces (links) are stable, we'll update the library. The changes will use the same interface locations for Black+RC, Blue and AI+RC. cc @lorforlinux

DhruvaG2000, 2021-04-02:
Hello! I was just going through [this](https://github.com/lorforlinux/bb.org-overlays) repo which has supposedly all the BBAI Overlays, done as part of adding Cape support for BBAI. However, after going through the entire repo I was unable to find the matching overlay for I2C other than i2c4. This could give rise to the above mentioned issues, as the DT Overlay for the rest of I2C ports is yet not supported. 
I could be wrong, but do you think this is what the issue could be @jadonk ?

lorforlinux, 2021-04-08:
@DhruvaG2000 the location of overlays has been changed during the `Cape Compatibility` project and all the overlays can now be found [here](https://github.com/beagleboard/BeagleBoard-DeviceTrees/tree/v4.19.x-ti-overlays/src/arm/overlays). You are supposed to submit all the new overlays there only :)

DhruvaG2000, 2021-04-10:
@jadonk I would like to contribute inorder to bring support for the BBAI+RC, Please can you assign this issue to me?

jadonk, 2021-04-13:
You don't need to be assigned to work on it. Take a look at my draft pull request and look for gaps that need to be filled. You can make your own pull request and pull in some or all of my changes thusfar.

jadonk, 2021-04-15:
Just helping to add some clarifications where issues come from...

> Update: I found am5729-beagleboneai-roboticscape.dtb and updated the configure_robotics_dt.sh script with a few lines of code:
> 
> ```
> TREE_BBAI_RC=am5729-beagleboneai-roboticscape.dtb
> 
> # test for BBAI
> elif   [ "$MODEL" == "BeagleBoard.org BeagleBone AI" ]; then
> 
> 	# if the roboticscape tree is available, use that
> 	if [ -a "/boot/dtbs/$UNAME/$TREE_BBAI_RC" ]; then
> 		DTB="$TREE_BBAI_RC"
> 	else
> 		echo "ERROR, can't find $TREE_BBAI_RC for this kernel."
> 		echo "no changes made to uEnv.txt"
> 	fi
> ```
> 
> This fixed the issue with some of the drivers, however the PWM, eQEP, and i2c still don't work:
> 
> debian@beaglebone:~/src/librobotcontrol/examples/bin$ rc_test_drivers
> 
> Kernel: 4.14.108-ti-r136
> BeagleBoard.org Debian Buster IoT TIDL Image 2020-04-06
> Debian: 10.4
> 
> PASSED: gpio 0
> PASSED: gpio 1
> PASSED: gpio 2
> PASSED: gpio 3
> ERROR: ti-pwm driver not loaded for hrpwm0
> ERROR: ti-pwm driver not loaded for hrpwm1
> ERROR: ti-pwm driver not loaded for hrpwm2
> ERROR: ti-eqep driver not loaded for eqep0
> ERROR: ti-eqep driver not loaded for eqep1
> ERROR: ti-eqep driver not loaded for eqep2

With different kernel versions as well as different SoCs, the PWM drivers have changed and the locations of the symlinks have changed as well. What exactly to do here needs a bit of thought, but `rc_test_drivers` correctly indicates it is not able to find the drivers for the desired PWMs.

A good start is to look at the PWM entry in the [Cape Interface Specification](https://elinux.org/Beagleboard:BeagleBone_cape_interface_spec#PWM) and switch the code to use that when it is available. Some additional symlinks and possible interface fixes may be required to be added to the interface spec and librobotcontrol.

> PASSED: pru-rproc
> PASSED: uart1
> PASSED: uart2
> PASSED: uart4
> PASSED: uart5
> ERROR: i2c1 driver not loaded
> ERROR: i2c2 driver not loaded

This is simply a matter of the indexes changing between AM3358 and AM5729 for the I2C buses connected to the right devices.

With the [Cape Interface Specification](https://elinux.org/Beagleboard:BeagleBone_cape_interface_spec#I2C), the new locations to find these bus drivers should be clear.

> PASSED: spi
> PASSED: LED
> PASSED: ADC iio
> 
> Currently running on a:
> MODEL_BB_AI
> Robot Control library Version:
> 1.1.0

DhruvaG2000, 2021-04-15:
I have verified [BeagleBone_cape_interface_spec#I2C](https://elinux.org/Beagleboard:BeagleBone_cape_interface_spec#I2C) many many times and the bone bus is pointing to the correct locations! Hence if the issue does not lie there, I fail to understand where else... 
The bone buses are configured already in the ``cape_interface_spec`` in the same way as was in the BBB and to the same pins as well, so as far as liborobotcontrol is concerned, the I2C bone bus should just appear the same to it be it the BBAI or BBB. Even inspite of this its showing errors? 
I felt there is no qn of changing indices since the cape interface has already done the job? 

/dev/bone/i2c/1  that pointed to I2C1 in BBB automatically should point to I2C5 in BBAI as per the overlay in place. Similarly for rest of the pins as well

OscarJGV26, 2022-07-03:
Hi all

Just wanted to check whether the Beaglebone AI is currently working with the Robotics Cape? Didn't get the picture of whether this is a solved problem or not.

Thanks a lot!

OscarJGV26, 2022-07-29:
Hi there

I just received my BB-AI with my robotics-cape. I was hoping someone could help me out with some problems I have with it?

OscarJGV26, 2022-07-30:
> @DhruvaG2000 the location of overlays has been changed during the `Cape Compatibility` project and all the overlays can now be found [...]

Can you explain how to use the overlays... I don't quite understand their general purpose as I don't see them inside the am5729-beagleboneai-roboticscape.dts or am5729-beagleboneai.dts files

jstumfoll2, 2022-11-19:
@OscarJGV26 Not sure if you're still having troubles. I gave up on my project a while back. I was hoping for a bit more development work to happen to fix some of my issues. I've recently picked it back up and looking into everything again. Looks like they did do a little bit to solve the BBAI problems, but the librobotcontrol library will still take some troubleshooting and problem solving for yourself to figure out. 

I would start by checking out this page: https://beagleboard.org/blog/2022-03-31-device-tree-supporting-similar-boards-the-beaglebone-example

and this page: https://elinux.org/Beagleboard:BeagleBone_cape_interface_spec

It seems they came up with a better way to handle the differences in the processor connections between the BBB and the BBAI. From my research though it seems that this is only implemented in the latest kernel builds, and they never updated the BBAI stable image. The first page gives a process for how to update the kernel

Now, I tried this kernel update and it borked my BBAI, it wouldn't boot. I had to change the udev file to revert to the old kernel, but YMMV. So, I don't actually know how the new updates work to fix the pin mapping/device tree overlays/etc. I will say that I've managed to get most things working on my own. The first page I linked gives a pretty good overview of the actual issue, and Jason's feedback above helps explain the process to fix everything as well. At some point I may try to clean up everything I've done to the librobotcontrol library to put it out there for anyone else.

I'm working on figure out the encoder PRU piece now. I just hit this line of code last night. https://github.com/beagleboard/librobotcontrol/blob/v1.1/library/src/pru/encoder_pru.c#L28

If anyone has any suggestions for what to do to fix the BBAI Encoder PRU let me know :)