beagleboard/librobotcontrol #175: installing via apt-get with -y option

athompson673, 2020-01-14:
Is there some way on installation or update to detect if `apt-get` was called with -y and not bring up a config menu, but rather take a default configuration? this is a bit of a pain when letting `apt-get -y upgrade` run overnight only to find a config menu open and waiting for input, when I explicitly wanted to give no input for the command.

RobertCNelson, 2020-01-14:
@athompson673 how old is your image? I haven't pushed a deb update to the apt repo for this package in over 10 months:

https://github.com/rcn-ee/repos/tree/master/git-librobotcontrol

Regards,

DhruvaG2000, 2021-04-10:
@athompson673 If your issue has been solved kindly close this issue