beagleboard/librobotcontrol #177: Encoder 0 returny only -1, 0 or 1

mdlugosz, 2020-04-10:
Hello, 
**Describe the bug**
Encoder E1 do not work propertly, return only values -1, 0 or 1. Tested using rc_test_encoders_eqep  software. If I switch encoder output to the another encoder input in BBB (E2, E3 or E4) everything works ok. 

**To Reproduce**
1. Install system using image bone-ubuntu-18.04.4-ros-iot-armhf-2020-03-16-6gb.img.xz from (https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Buster_IOT_Snapshot) 
This image contains also ROS Melodic and preinstaled librobotcontrol package.
2. Connect motor encoder to E1 BBB encoder input.
3. Run rc_test_encoders_eqep 
4. Move maunaly shaft of motor and see program results on screen for E1 encoder
5. Encoder E1 do not work propertly, return only values -1, 0 or 1 (depend on which direction of rotation of shaft motor)

**Expected behavior**
If I switch to the another encoder input (E2, E3 or E4) everything works ok so E1 should also work in the same way.
If I use another image for BBB (ex. AM3358 Debian 10.3 2020-04-06 4GB SD IoT - from official site) E1 works ok - so the problem is in software not hardware. The library Adafruit Beaglebone IO also work in the same way (E1 work incorrectly, rest encodes E2, E3, E4 work ok)

**Platform information**
Command lsb_release -a return:
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

rc_test_drivers 
Kernel: 4.19.94-ti-rt-r37
BeagleBoard.org ROS Image 2020-03-16
Debian: buster/sid

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

arvrml, 2020-05-14:
**Additional Bug Comment**
1) rc_test_encoder_eqep
I have the same issue with  Encoder E1 only outputting -1,0,1.  E2 and E3 are fine.

2) E4: is running from the PRU requiring sudo access but if fails as shown below

debian@beaglebone:/usr/bin$ sudo rc_test_encoders_pru
[sudo] password for debian:
ERROR in rc_encoder_pru_init, am335x-pru0-rc-encoder-fw failed to load
attempting to stop PRU0
ERROR: failed to run rc_encoder_pru_init

**Platform Information**
debian@beaglebone:~$ rc_test_drivers

Kernel: 4.19.94-ti-r42
BeagleBoard.org Debian Buster IoT Image 2020-04-06
Debian: 10.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4

clarkbriggs, 2020-05-20:
I have same issue with the first encoder (encoder 1). Other 2 eqep encoders work ok.
My set up: 
rc_test_drivers
Kernel: 4.19.94-ti-r42
BeagleBoard.org Debian Buster IoT Image 2020-04-06
Debian: 10.3
PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio
Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4
In addition, there is a complaint in dmesg after finding and configuring all 3 eqeps
[    1.118230] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.129640] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.137227] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
[    1.149602] bone-pinmux-helper ocp:P9_30_pinmux: Error applying setting, reverse things back
[    1.158197] bone-pinmux-helper: probe of ocp:P9_30_pinmux failed with error -22

After groping around for a long time, it looks like this is a conflict somehow between encoder 1 and an SPI channel which uses P9_30. P9_30 is apparently SPI1_MOSI P9_30 D12 SPI1_D1 GPIO3_15.
encoder 1 uses QEP_1A  B12  EQEP0A_IN  GPIO3_18 and QEP_1B P9_27 C13 EQEP0B_IN GPIO3_19. This leaves P9_30_pinmux broken with no "state" access. I feel bad for SPI1, but why does this mangle enocder 1? Is this conflict the problem?

qwert75, 2020-10-19:
Same issue exact same setup as clarkbriggs, any workarounds found?

clarkbriggs, 2020-10-23:
No fix that I have seen. The work around for me was to not use encoder 0. I only need two, so...ok. But you are right. Its been awhile hasn't it.

qwert75, 2020-10-23:
My work around was to use an older debian image, 9.9 i think. still has 1.0.4 robot control library but doesn't suffer the same issue.