beagleboard/librobotcontrol #180: PRU1 or PRU2?

clarkbriggs, 2020-04-30:
**Describe the bug**
rc_test_servos fails to start PRU1

**To Reproduce**
Steps to reproduce the behavior:
run sudo rc_test_servos -p 0.0
get gripe
ERROR in rc_pru_stop while writing to remoteproc state: Operation not permitted
ERROR in rc_servo_init, failed to start PRU1

**Expected behavior**
Successful completion and servo motion, but no servo installed.

**Platform information**
Blue purchased Aug 2019. Current image AM3358 Debian 10.3 2020-04-06 4GB SD IoT.
uname -a
Linux beaglebone 4.19.94-ti-r42 #1buster SMP PREEMPT Tue Mar 31 19:38:29 UTC 2020 armv7l GNU/Linux
rc_version
1.0.4
rc_test_drivers

Kernel: 4.19.94-ti-r42
BeagleBoard.org Debian Buster IoT Image 2020-04-06
Debian: 10.3

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4
Discussion:
Chasing rc_servo_init, rc_pru_stop, rc_pru_start, servo.c chooses PRU1:
#define SERVO_PRU_CH	1	// PRU1
and starts it
if(rc_pru_start(SERVO_PRU_CH, SERVO_PRU_FW)){...
In pru.c, PRU1 is associated with remoteproc2:
#define PRU1_STATE	"/sys/class/remoteproc/remoteproc2/state"
On my machine, there are 3 remoteprocs: 0, 1, 2. 
debian@beaglebone:/sys/class/remoteproc/remoteproc0$ cat name
4a334000.pru
debian@beaglebone:/sys/class/remoteproc/remoteproc1$ cat name
4a338000.pru
debian@beaglebone:/sys/class/remoteproc/remoteproc2$ cat name
wkup_m3
debian@beaglebone:/sys/class/remoteproc/remoteproc2$ cat state
running
Is robotcontrol supposed to be using remoteproc 0 & 1? Isn't something else using remoteproc2?
Is this a kernel enumeration issue (like of olden days)?
I think this configuration is prety close to out-of-the-box, but servos don't pass test.
Please advise.
Clark

RobertCNelson, 2020-04-30:
As of last week, we do have a better udev rule to fix this order issue:

https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/buster/debian/86-remoteproc-noroot.rules#L21-L25

```
apt update / apt upgrade
```

to get it..

Regards,

StrawsonDesign, 2020-04-30:
Thanks Robert!

clarkbriggs, 2020-05-01:
Robert, Wow! timely, no? But...
I can't comprehend what the udev rule does but it didn't change the order of the remoteprocs on my Blue. I did as you said above to upgrade and I did see bb-customizations come along.
dmesg contains
debian@beaglebone:/sys/class/remoteproc/remoteproc2$ dmesg |grep pru
[    8.750811] remoteproc remoteproc0: 4a334000.pru is available
[    8.751000] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    8.772797] remoteproc remoteproc1: 4a338000.pru is available
[    8.773004] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
debian@beaglebone:/sys/class/remoteproc/remoteproc2$ dmesg |grep remoteproc
[    8.750811] remoteproc remoteproc0: 4a334000.pru is available
[    8.772797] remoteproc remoteproc1: 4a338000.pru is available
[   44.729954] remoteproc remoteproc2: wkup_m3 is available
[   45.852253] remoteproc remoteproc2: powering up wkup_m3
[   45.852288] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[   45.852582] remoteproc remoteproc2: remote processor wkup_m3 is now up
and rc_test_servos still fails the same way
debian@beaglebone:~$ sudo rc_test_servos -p 0.0
ERROR in rc_pru_stop while writing to remoteproc state: Operation not permitted
ERROR in rc_servo_init, failed to start PRU1
hmmm,
Clark

RobertCNelson, 2020-05-01:
@clarkbriggs it's more of.. Yes, it's a problem, we (well zmatt) figured out a good solution...

But... Every library needs to be updated to use the "helpers". ;)

So instead of:
```
#define PRU1_STATE "/sys/class/remoteproc/remoteproc2/state"
```

You'd write:
```
#define PRU1_STATE "/dev/remoteproc/pruss-core1/state"
```

And never worry about it changing again!  But sadly, we are usually more reactive then pro-active with these issues.. Thus we have a "fix" but it came last week, and not 1~2 years ago so this library would have had the ability to use it. ;)

Regards,

RobertCNelson, 2020-05-01:
I should mention this probably is also an issue with v4.14.x-ti, as of the same time frame... earlier this month when i was merging all TI's TIDL changes for AM57xx devices, omap_remoteproc had to be moved from built-in to module.. So on the am335x, the m3 power management core, is now in a boot init race with pru0 and pru1 for "remoteprocX" values... (that's the 3 remoteproc directories..)

Regards,

clarkbriggs, 2020-05-02:
James,
I don't know what Robert said beyond "yeah, its broke", but I have got to get on.
So instead of using your hardwired paths to a remoteproc for each PRU, I codefully grunge through all the remoteproc dirs looking at firmware names to identify which remoteproc supported which PRU. Then I did all the usual PRU state manipulation through that remoteproc dir. It seems like it would be robust to the init enumeration order, but I can't test it.
I got this to work for the servos and haven't tried it for the encoders yet.
TTFN. OBTW, your library is VERY nicely done.
Clark

ConnorChristie, 2020-06-18:
So what is the workaround for this? Or is there a formal solution yet?
I am running a beaglebone blue with the latest kernel `Linux beaglebone 4.19.94-ti-rt-r44 #1buster SMP PREEMPT RT Mon Jun 8 20:23:42 UTC 2020 armv7l GNU/Linux` and `rc_version 1.0.4` and it's giving me the same errors...

@clarkbriggs Would you happen to have a set of the commands you used to get this working?

jadonk, 2020-06-18:
@RobertCNelson Is this one of those cases where I should make a hack point release that uses the new names and require folks to install/update bb-customizations to make it work? Does the librobotcontrol package depend on bb-customizations already? I think @StrawsonDesign was a little more direct in his updates, but if we have people who are just stuck, it seems we need to do something sooner than later.

I should at least sleep on it to try to figure out how to release test packages to those not building from source.

clarkbriggs, 2020-06-18:
ConnorChristie asked if I have commands that work. I did what I wrote on May 2. Its code, not command line commands. The biggest disadvantage is once you start hacking StrawsonDesign's code, its hard to find the boundary. Of course, forking it would be the correct way. I have been making my own mangled stand alone implementation following his design, with his consent. My approach was to search each remoteproc dir looking at the string returned from reading firmware and using string search to see if it mentioned the desired PRU. Its not necessarily robust, for example, I use a static list of the three remoteproc dirs. But it got me past it and I went on to other issues. A serious downside is that my code works but the distro code doesn't, in those cases that require the PRUs. For example, I went on to an encoder issue that I have and have to use rc_test_encoders_eqep, not rc_test_encoders. But that work around was ok. There isn't such a work around for the servos, so I can't use the distro rc_test_servos.

jadonk, 2020-06-18:
Please test https://github.com/beagleboard/librobotcontrol/releases/tag/V1.0.5-RC1

ConnorChristie, 2020-06-19:
Thank you all for the extremely quick responses! @jadonk I just tried your changes with the latest PX4 firmware and it works a treat. No more "Operation not permitted" during startup :)

Thank you again for the extremely quick turnaround.

jadonk, 2020-06-22:
Did a quick, but not thorough test on a PocketBeagle. Really only tested a single alias.

```
debian@pbgl00a20125:/sys$ rc_test_drivers 

Kernel: 4.19.94-ti-r42
BeagleBoard.org Debian Buster IoT Image 2020-04-06
Debian: 10.3

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
ERROR:  uart5 driver not loaded
PASSED: i2c1
PASSED: i2c2
ERROR: spi driver not loaded
ERROR:  LED driver not loaded
PASSED: ADC iio

Currently running on a:
MODEL_BB_POCKET
Robot Control library Version:
1.0.4
```

clarkbriggs, 2020-07-07:
jadonk: re test v1.0.5-RC1. Done. Seems to work. That is rc_test_servos does not fail to open the PRU. Nice work. Thanks.
Looking at the src for pru.c in v1.0.5-RC1, I can see what RobertCNelson meant on Apr 30. Pretty straight forward. I'll do it that way instead of my search all 3 remoteproc dirs as I wrote on May 2. Thanks.
I would guess this addresses my issue. Should it be closed? Who does that? I'm just going to post this with the "Comment" button because I don't know the right github process here.

closed 2020-07-16 by jadonk