beagleboard/librobotcontrol #188: Unable to run rc_test_servos on latest image release.

ansarid, 2020-07-05:
When trying to run `rc_test_servos` on the latest beaglebone image release (AM3358 Debian 10.3 2020-04-06 4GB SD IoT) on a Beaglebone Blue, the error below is returned.
```
debian@beaglebone:~$ sudo rc_test_servos -c 1 -p 0.0
ERROR in rc_pru_stop while writing to remoteproc state: Operation not permitted
ERROR in rc_servo_init, failed to start PRU1
```

jadonk, 2020-07-16:
Can you try the V1.0.5-RC2 release?

DhruvaG2000, 2021-04-13:
Hey @ansarid was your issue solved or is there any new errors you are facing? 
Kindly close this issue if your issue has been solved

ansarid, 2021-04-13:
@DhruvaG2000 @jadonk The issue no longer exists, and I forgot about this issue, sorry!

closed 2021-04-13 by ansarid

pallantirao, 2021-12-11:
Hi, I had the same issue as shown below. But the scenario is as follows: Had three boards one with older version ad other two with 1.0.5 version updated. The older version has -v option enabled where as the 1.0.5 version is removed and seem to have hard coded. the two biards with 1.0.5 version , the same procedure followed for updating OS and librobo. But one board has no issue but anither board failed giving the the following messgae. Great if some one helps.
debian@beaglebone:~$ sudo rc_test_servos -c 1 -p 0.0
ERROR in rc_pru_stop while writing to remoteproc state: Operation not permitted
ERROR in rc_servo_init, failed to start PRU1