beagleboard/librobotcontrol #192: Webcam loses power when using rc_test_motors example

ninjatrent, 2020-08-24:
Logitech C920 webcam loses power and mjpg-streamer stops working when using the rc_test_motors example. Camera is active on boot and works when using " rc_test_motors -d 0.5 " but loses power when motors are reversed with " rc_test_motors -d -0.5 ". 

It seems to be an issue with power in the USB bus. When a lower duty cycle is specified ( -d -0.3), the camera remains powered. Anything more and the video stream freezes and cam loses power. It's strange that this didn't occur in previous use.

I'm using the Logitech C920 webcam with the BeagleBone Black W and the Robotics Cape on the DFRobot Devastator Robot Chassis.

silver2row, 2020-10-04:
Hello @ninjatrent ,

I understand things at times. Did you plug in the USB to a power bank while still powered via LiPo or have you used the 12v barrel jack supply too?

Seth

P.S. I am just trying to learn the source they have provided, i.e. so I might not know exactly what is causing the issue outside of a power concern. Off to read the source and where that source was derived!

StrawsonDesign, 2020-10-04:
What is your power source? Sounds like it can't keep up with the current
in-rush.

ninjatrent, 2020-10-18:
Hi Seth,

The BB is receiving power from a LiPo battery. The usb webcam is connected to the usb on the BB. The cam is likely drawing more power than the usb port can supply when the motor drivers are at 75% + duty. I might take the Logitech webcam apart and disable the blue leds on either side of the lens. This might alleviate the hiccup.

closed 2020-10-18 by ninjatrent