beagleboard/librobotcontrol #194: RC binding and esc calibration loss

YevheniiKovryzhenko, 2020-08-27:
1. Whenever I power on the Beaglebone blue I have to rebind my Spektrum DX6e radio control every time. Moreover, when I have completed the binding procedure (RC works, checked with rc_test_dsm), if I power down the RC I have to rebind again. I have checked all the binding options available and for those that worked (I am using option 3) the same issue was present. I am using Spektrum DSMX receiver connected to UART DSM port. 

2. Similarly to the first issue, one or multiple of the ESCs randomly tend to stop responding to any commands. I haven't been able to determine what causes this issue, and the only way to fix it is to run rc_calibrate_escs. It seems to be very random: I might need to run re-calibration only once in a few days, sometimes a few times in an hour. It looks like this issue is independent of rebooting or powering on/off entire setup and I am not rewiring anything before or after rebinding, so I don't think it's the wiring issue.

Also, I had the same issue on different BeagleBone Blue boards, so it shouldn't be the hardware problem (unless I got extremely unlucky).

I don't have a lot of experience with programming in C/C++, so this is as much as I was able to figure out for now.

I have updated the libraries to the latest version - both issues are still present.   

Any and all help will be much appreciated!

StrawsonDesign, 2020-08-28:
Sorry, I know this doesn't answer your questions fully but I just wanted to
chime in with some info.

Some Spektrum RC receivers, including the "FPV" receivers with the long
antenna, are designed to lose pairing if the receiver ever starts up
without the transmitter on. If this happens it has to go through the bind
process again regardless of if it's on a Beagle one/pixhawk/standalone.
Maybe try turning the transmitter on before the Beaglebone? I've tried many
receivers and have only seen the behavior you described with the FPV ones.

Regarding the ESCs, I've found some ESCs to exhibit this behavior even with
a Pixhawk flight controller and I always suspected random noise in the
signal line to trigger it. It's a pity you are seeing at random as the next
debug step would ideally be to put an O-scope on signal to see if there are
any weird behaviors during startup. I've ditched old fashioned ESCs because
of annoyances like this.

YevheniiKovryzhenko, 2020-08-31:
Thank you! I will try powering the transmitter first and see if that changes. I wasn't sure if this issue is software related of hardware. I might be able to test the other rc radio that I will be getting soon as well, and see if it works better. 

May I ask what are you using instead of the ESCs to control the motors? I am not aware of a different approach to controlling brushless motors.

Also, probably not related to these issues, but I have had problems getting Altitude estimation to give more or less stable values. After performing all the calibrations, if I run rc_test_altitude or rc_test_mpu I always end up with estimated altitude that oscillates around +-1m of some nominal (ex. 176m) value and it tends to stabilize for a few seconds, but after a second or two it jumps or drops again. I understand that state estimation using barometer is not ideal, but I thought it would be more or less stable when Kalman filter is incorporating IMU sensors and the altitude would not be oscillating as much as it does.

Thank you again for your work and time!

closed 2022-04-19 by YevheniiKovryzhenko