beagleboard/librobotcontrol #213: Debian 11 Updates and Porting/Working with librobotcontrol

silver2row, 2021-11-09:
**Describe the bug**
The install is not available with dpkg nor can I get the install completed with the correct ` rc_test_drivers ` command.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/beagleboard/librobotcontrol
2. cd librobotcontrol
3. make
4. sudo make install
5. rc_test_drivers

**Expected behavior**
I expected the drivers to be available and for my drivers for GPIO, PWM, UART, and SPI to pass.

**Screenshots**
```
debian@Me:~/librobotcontrol$ rc_test_drivers

Kernel: 5.10.65-ti-r24
BeagleBoard.org Debian Bullseye IoT Image 2021-10-10
Debian: 11.1

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm0
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
ERROR:  pru-rproc driver not loaded
ERROR:  uart1 driver not loaded
ERROR:  uart2 driver not loaded
ERROR:  uart4 driver not loaded
ERROR:  uart5 driver not loaded
ERROR:  i2c1 driver not loaded
PASSED: i2c2
ERROR: spi driver not loaded
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.5
```

**Platform information**
1. cat /etc/dogtag : ` BeagleBoard.org Debian Bullseye IoT Image 2021-10-10 `
2. uname -r : ` 5.10.65-ti-r24 `

** Additional Info. **
I have also been trying to ` repair ` rcpy and pycntrl to no avail. Anyway, I will stick w/ librobotcontrol as I know it has worked in the past. If anyone needs help updating old scripts or making current GCC-10 ideas available, please direct me.

silver2row, 2022-01-06:
Also,

This may seem silly but here goes it. On the OrangeRX R820X V2, there are a few JST-ZH ports with three-pin assignments. There are eight "channels" on it also. So, there are a row of eight w/ three pins each as columns. 

Anyway, I have come to a conclusion that I cannot access the receiver from and to the BBBlue. Is anyone using the librobotcontrol any longer w/ the BBBlue? I am basically looking for handouts here b/c I feel like one wrong turn and I could ruin the BBBlue. 

Seth

P.S. I cannot find the V2 "datasheet" any longer. But! I found a V3 versioning of the OrangeRX "datasheet". I have quoted the datasheet wording b/c of the lack of data in their sheets. If you are using the BBBlue w/ librobotcontrol and a DSM (UART) connection, please do let me know about what connections to make on the BBBlue to/from the receiver for DSM and PRU.

unl0ck, 2022-04-19:
I am using BBBlue with librototcontrol, but have issues... encoder 4 make trouble and no solution is present

silver2row, 2022-05-12:
@unl0ck : 

Hello...I think in the .dts file, the PRUs are used to handle the encoder on connector four. 

Seth

P.S. You can find it in the BBBlue .dts file called am335x-boneblue.dts.

unl0ck, 2022-05-12:
I am using it, but its still show 0 -1 1 not more do you want to see a special line?