beagleboard/librobotcontrol #221: [BBBlue] rc_test_drivers broken for kernel linux >=5.x.y.z

menschel, 2023-05-19:
Hello, I have reactivated a BeagleBoneBlue that was catching dust on my shelf and while updating the kernel.

via 

`sudo /opt/scripts/tools/update_kernel.sh --lts-5_4`

I found out that librobotcontrol seems broken on all kernels after 4.19.94-ti-r73.

```
Kernel: 5.4.106-ti-r41
BeagleBoard.org Debian Buster IoT Image 2020-04-06
Debian: 10.13

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm0
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
ERROR:  i2c1 driver not loaded
PASSED: i2c2
ERROR: spi driver not loaded
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.5

```

Is this a librobotcontrol issue or is this a DeviceTree issue?

acochrane, 2023-12-23:
I am also looking for a solution here.
Poking around, there's a AM335X-PRU-RPROC-4-19-TI-PRUCAPE-00A0.dtbo for kernel 4.19, but no dtbo for kernels higher.
This device tree stuff is pretty foreign to me, I'm not really getting far.

I saw this [forum post](https://forum.beagleboard.org/t/overlays-beaglebone-ai-64-pwm/32508) for BBAI PWMs. So there's source code for arm64, shouldn't be hard to move to arm.

Is it an overlay problem?

How do we make overlays for BeagleBone Blue?
How do we load ti-pwm drivers?

menschel, 2023-12-23:
Honestly speaking, librobotcontrol widely seems abandoned for good.
I will create a pull request to make motors and eqep running again later but the PRUs are a different story. Servo PWM does not work at all, maybe it's the custom PRU firmware used in librobotcontrol.

Some fixed are currently deployed for the devicetree. 
https://github.com/beagleboard/BeagleBoard-DeviceTrees/pull/69

necessary changes for librobotcontrol will be using other PATH, no unexport / export of chips and finally porting to use libgpiod for gpio control and pwm.h for pwm control.

This was all done in Kernel 5.x.y.z.

jjd35, 2024-05-11:
FYI, for anyone else who might be looking for how to use the PRU firmware within this repo for the later SD images (which have kernel versions >=5.x.y.z). You have to change the pinmux for the AM335x chip. The later device trees must've changed it. You can drop in a device tree overlay like this to change the pinmux to map to the same pins as the librobocontrol PRU Firmware, this only updates the pinmux such that PRU1 can output to the Servo pins of the beaglebone blue using the same R30 register bits as the firmware :
  ````
   #include <dt-bindings/pinctrl/am33xx.h>
   /dts-v1/;
   /plugin/;
  / {
	  compatible = "ti,am335x-bone-blue", "ti,am33xx";
      /* identification */
	  part-number = "PruServoOutput";
	  /* version */
	  version = "00A0";
      fragment@0 {
          target = <&am33xx_pinmux>;
          __overlay__ {
              pru1_svo_output_pins: pins {
                  pinctrl-single,pins = <
                  AM33XX_PADCONF(AM335X_PIN_LCD_VSYNC, 0, MUX_MODE5) /*pru1_pru_r30_8, MODE5*/
                  AM33XX_PADCONF(AM335X_PIN_LCD_PCLK, 0, MUX_MODE5)  /*pru1_pru_r30_10, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_HSYNC, 0, MUX_MODE5) /*pr1_pru1_pru_r30_9, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_AC_BIAS_EN, 0, MUX_MODE5) /*pru1_pru_r30_11, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_DATA6, 0, MUX_MODE5) /*pru1_pru_r30_6, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_DATA7, 0, MUX_MODE5) /*pru1_pru_r30_7, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_DATA4, 0, MUX_MODE5) /*pru1_pru_r30_4, MODE5 */
                  AM33XX_PADCONF(AM335X_PIN_LCD_DATA5, 0, MUX_MODE5) /* pru1_pru_r30_5 MODE5 */
                  >;
              };
          };
      };

    fragment@1 {
        target = <&ocp>;
            __overlay__ {
                pru1_svo: pru1_svo {
                compatible = "bone-pinmux-helper";
                pinctrl-names = "default";
                pinctrl-0 = <&pru1_svo_output_pins>;
                status = "okay";
            };
        };
    };
};