beagleboard/librobotcontrol #230: Readme.md link to documentation leads to 404 page

dotbit1, 2024-09-11:
**Describe the bug**
http://beagleboard.org/librobotcontrol/ linked form Readme.md in root leads to a 404 page.

jadonk, 2025-01-06:
I'll need to make sure bug fixes for this going into https://openbeagle.org/beagleboard/librobotcontrol are being propagated to here. Thank you for reporting the issue and sorry for the delayed response. Someone mentioned this on the Discord and got me going to fix this.

jadonk, 2025-01-06:
In the meantime, the staged fix is at https://jkridner.beagleboard.io/docs/projects/librobotcontrol/index.html and will be moved soon to the final destination once reviewed.

jadonk, 2025-01-06:
fyi, also an auto-generated version at https://beagleboard.beagleboard.io/librobotcontrol/index.html